reductus/reductus #4: caching needs small overhaul

bmaranville, 2016-04-14:
Can't cache offspecular data type using JSON, as was done for refl files.  Nested self-describing dicts with embedded datatype just won't work.  

The serialization of ospec data gives a binary string - need to be able to make bundles of that.

Maybe we go back to structured storage (rpush and lrange?)

pkienzle, 2016-04-14:
The use of JSON vs. some binary format seems to be orthogonal to the
rpush/lrange question.  Or do I misunderstand the question?

bmaranville, 2016-04-14:
Not orthogonal... Bundles seem to work because lists pop right into/out 
of JSON structures.  If you are going to store data in a binary string, 
and can't use JSON anymore, the trick of throwing the whole structure 
into the serialization doesn't work (goes all the way up to the node 
level right now: {"terminal_id1": {"datatype": "refldata", "values": 
[<serialized item>, <serialized item>,...]}, "terminal_id2": ...})
Caching the list at the bottom of that chain becomes the problem if each 
data object is a binary string, which is why it's connected to the 
lrange/rpush question.

bmaranville, 2016-04-14:
I found a library called ujson that allows non-ascii (binary!) strings in JSON encodings.  Also looking into BSON as a dropin replacement for JSON for this.

pkienzle, 2016-04-14:
Changed to pickle for redis store.

closed 2016-04-14 by pkienzle