reductus/reductus #5: better handling of zero and negative values on log-scale axes

pkienzle, 2016-05-09:
Present negative data as log(abs(y)) on an inverted scale below the normal log scale graph.   Furthermore, set the yscale minimum to min(abs(y)+dy)/2 to protect against zero counts in the data.  This will be useful for specular and background subtracted data.

For spin asymmetry data a linear scale would be better even if yscale is set to log.  Maybe the plot info could carry a flag indicating that yscale should be set by the data rather than the UI?

bmaranville, 2016-05-17:
Plotting negative data on a log scale plot is likely to be confusing to people (most?) who aren't familiar with the practice.  

I'm happy to use the 'ytransform' from the plot options, though - if it's present I'll use it, otherwise things work as they do now.

bmaranville, 2016-05-17:
Piping through the xtransform and ytransform is implemented and used with spin asymmetry - see d05794d

closed 2016-05-17 by bmaranville