reductus/reductus #9: feature request: join Qz_target when present

bmaranville, 2016-05-28:
Please also do a join operation on the Qz_target column when joining, so that in the future we can move to a workflow where the alignment of the data is more explicitly defined.

pkienzle, 2016-12-16:
If Qtol and dQtol are 0 then the desired angles are used for join instead of the actual angles.  Does this satisfy the request?

bmaranville, 2016-12-21:
This seems kind of non-intuitive.  Why not make it explicit instead of 
implicit?

pkienzle, 2016-12-21:
That works too.  Increased user interface clutter but perhaps more obvious to the user?

If not using Qz_target, and Qtol and dQtol are 0, does this mean that points will only be combined if their actual motor positions are identical?  Is this useful?

For candor we may want the option to combine by Q across lambda, so that will be yet another option on this screen; still want target vs. actual for this I suppose so maybe it doesn't affect the UI but I didn't want to make that decision at the time, so I went with the minimal user interface change.

Regarding join by Q, I've got code for it already but it won't work since the refldata structure can't handle it.  It computes Qz from angles and lambda rather than using a stored value.  Making the refldata structure dumber and moving more work into the calculation components is probably the way forward.  I'd prefer to put off any major restructuring until we incorporate more instruments, and in particular, the TAS instruments which have more complicated join logic.

pkienzle, 2020-03-10:
We've had a *Qz_target* as an option for years now, so closing this ticket:

    2015-11-24 dbfe6c0 "adding Qz_target to dataset" bmaranville

closed 2020-03-10 by pkienzle