reductus/reductus #11: long filenames cut off in legend

pkienzle, 2016-08-19:
the file names listed in the graph legend get cut off if the filename is too long.  This makes it difficult to figure out which file is associated with which line in multi-file selections, since they are usually distinguished by sequence number at the end rather than sample condition at the beginning.

Choosing a smaller font on the browser works (e.g., using Ctrl-Minus).  The solution may be to document this rather than changing anything.

Another option would be to make the legend movable, which you may want anyway if the legend is covering some of the data.

bmaranville, 2017-03-17:
Adding an option to pop out the legend to outside the graph (resize graph to make room on the right?) might address this issue when there are large numbers of things plotted at the same time.

bmaranville, 2017-03-30:
Legends movable now: see https://github.com/bmaranville/d3-science/commit/2df4f819fc9df24b74a508b424f06ed52860626a

closed 2017-03-30 by bmaranville