reductus/reductus #21: data highlight based on legend doesn't work for single point scans

pkienzle, 2016-12-17:
The highlight on hover when moving the mouse over the graph legend only emphasizes the line thickness, it does not emphasize the points, so it does not work for single point slit scans.

Ideally, the target line would be raised above the rest of the data without thickening the line (so that error bars are still visible), and the remaining lines would be faded so it is easier to see the prominent line.  Reflpak also made the highlighted line red.  A possibility is to redraw the line above the rest.

bmaranville, 2016-12-21:
Highlight currently doesn't alter the drawing directly (just changes CSS) but it's something that could be done.  It's more complicated to store the original drawing, draw it with the new line on top, then redraw the original when highlight is removed.  Reording lines can't be done with just CSS as far as I can tell (I tried)

pkienzle, 2016-12-22:
Can you reduce opacity on the non-highlighted points and lines?  That would
allow the underlying line to show through in most cases.  Unfortunately,
this feature is most needed when there are the largest number of lines
together :-/

Haven't looked at your css, but assuming you have a class for data lines on
the graph, and an additional class for the highlighted line, then something
like the following:

    .dataline { filter: opacity(0.2); }
    .dataline.highlighted { filter: opacity(1); }

Same for data points.

Seems to work on safari/chrome/firefox.  Haven't tried it on IE.

The other thing I did in reflpak was reserve the red color for highlighted
points and lines, but probably not necessary if you can easily control
opacity of all the other lines.

  - Paul

bmaranville, 2017-03-17:
I couldn't get filter: opacity(x) to work at all... did you mean filter: alpha(opacity=0.2)?  
In any case I think the recommended css these days is just opacity: 0.2;

commit 5958523f (2017-03-17): switching to the new ES2015 xyChart library, which has unhighlight feature; Fixes #21
closed 2017-03-22 by bmaranville