reductus/reductus #22: default export file based on the name of the dataset

pkienzle, 2016-12-17:
Names of the datasets are preserved by the reduction process.  Join takes the first name in the dataset, which ideal is sorted by filename, so it will correspond to the earliest file.  Subtraction and slit normalization will use the name of the specular, so the name should be predictable and useful.

pkienzle, 2016-12-17:
Consider using separate extensions for the different components so they aren't all called refl files; probably not worthwhile.

bmaranville, 2017-01-31:
We can come up with a different suffix if you like, but I don't think it will help in the long term to multiply the number of suffixes used (want users to be able to identify reloadable exports by suffix).  Possibly one suffix per instrument?

pkienzle, 2017-01-31:
I would like the refldata base name to be the default base name for export, so that specular files BiFeO3-012.nxz.cgd through BiFeO3-016.nxz.cgd default to an export name of BiFeO3-012.dat.

Regarding the names for the intermediate steps, having BiFeO3-012.spec.dat for the specular files before background and slit correction is probably not worthwhile since the spec file will almost never be saved, and the user can easily override the name if they need it.

bmaranville, 2017-01-31:
Worry not, the original ticket will still be addressed.  The first addendum is the thing I think we shouldn't do.  It's a little more complicated to tackle this than I'd like, because the export function at the moment is passing through text strings from the server so the editor doesn't know the "name" attribute without parsing the text, which is a terrible idea.

bmaranville, 2017-01-31:
There is also a use case that has come up already (parallel data reduction of many data sets) for an exploded export (zip file full of individual exports, each named something unique), which I am trying to facilitate, or at least not prevent, with this change.

bmaranville, 2017-02-01:
This is fixed as part of https://github.com/reflectometry/reduction/tree/bcb0af4365249a56d2cd10ea949f515567e3d7ca

bmaranville, 2017-02-01:
Fixed in bcb0af4365249a56d2cd10ea949f515567e3d7ca which is part of new feature completed in 5e06e7eb7b4def9395040ccc7c169077a04d7e65

closed 2017-02-01 by bmaranville