reductus/reductus #29: redo footprint correction so footprint is plotted

pkienzle, 2017-02-16:
The "footprint correction" module currently accepts a slope-intercept from a fitted footprint module. Instead, it should accept a footprint curve in with correction values for each point in the dataset.  This would allow the user to see the footprint before applying it, which makes it easier to adjust the footprint parameters.  Use various footprint modules (ab initio, slope-intercept, measured) to set the scaling value for each point.  Note: both the ab initio and the slope-intercept models have fittable parameters, and so they should include a "fit" flag in the options.

Slope-intercept should not be fit beyond the fixed slit region.  It would be convenient to tag the fixed-slit region on the graph, or otherwise make it easy to discover.

bmaranville, 2017-02-16:
The same result in this case could also be accomplished by putting some interactors on the graph, with a line for the footprint correction and a range interactor for the applied region.  We don't have a generic way of adding per-module graph interactors though - up to now all the interactors are based on field input types.  I suppose "line" and "range" implementations for field types would accomplish this, possibly breaking old reductions.

bmaranville, 2019-07-30:
Probably a more sensible solution would be to allow plottable objects from the server side to contain specifiers for overlays (not just xy data or 2d data to plot).  This would also help with showing masks for SANS data.