reductus/reductus #39: export in py3/hug is broken with proxy

bmaranville, 2017-05-10:
when using uwsgi with a proxy balancer, export does not work (refldata.ncnr.nist.gov)

bmaranville, 2017-05-10:
```
in __call__
    return __hug_internal_self._function(*args, **kwargs)
  File "./api.py", line 193, in calc_terminal
    to_export = retval.get_export()
  File "/home/bbm/reduction/dataflow/core.py", line 507, in get_export
    values = [v.export() for v in self.values]
  File "/home/bbm/reduction/dataflow/core.py", line 507, in <listcomp>
    values = [v.export() for v in self.values]
  File "/home/bbm/reduction/reflred/refldata.py", line 927, in export
    _write_key_value(fid, n, getattr(self, n))
  File "/home/bbm/reduction/reflred/refldata.py", line 954, in _write_key_value
    fid.write(b'# "%s": %s\n'%(key.encode('utf-8'), value_str.encode('utf-8')))
TypeError: unsupported operand type(s) for %: 'bytes' and 'tuple'

```

bmaranville, 2017-05-10:
This is on python3.4, btw (ubuntu 14.04)

bmaranville, 2017-05-10:
Seems to be fixed by 081b9ec3e18806ae7d30e204939251f72d0cf162

bmaranville, 2017-05-10:
After some searching, I found that % formatting for byte representations was not introduced until python 3.5

pkienzle, 2017-05-10:
Do we need to support 3.4?

bmaranville, 2017-05-11:
The VMs provided by Alan tend to be ubuntu 14.04 LTS, which has that 
version of python. (in particular refldata.ncnr.nist.gov, which is our 
main internal server for the app...)

bmaranville, 2017-05-11:
I can install 3.5 from source on the VM, but for a production machine I 
would prefer to use a standardised setup.  Speaking of which, I need to 
add the apache configs and some startup scripts to the repo.

closed 2017-05-15 by bmaranville