reductus/reductus #54: need matrix multiply and invert error propagation in Uncertainty class

bmaranville, 2019-07-16:
@pkienzle :
For polarization correction in SANS, they need to be able to propagate errors through matrix multiply and inversion (looks like this is NotImplemented in dataflow.lib.uncertainty.Uncertainty)

bmaranville, 2019-07-17:
I guess there is already this functionality in uncertainties.unumpy package... I am less familiar with the specifics of using Uncertainty vs. uncertainties.

pkienzle, 2020-03-11:
Uncertainty is faster but uncertainties is more correct.

In particular, uncertainties will maintain correlated error calculations through entire expressions while Uncertainty treats each step as independent random variables.  So if you start with a = 1 +/- 0.01, then (a+1)/a has uncertainty 0.022 with Uncertainty, but 0.01 with uncertainties.  Using monte carlo, I get 0.01.

For a large array (e.g., 512x608), it takes 2.2 s to construct and 4.2 s to add a constant with uncertainties, but with Uncertainty this is 470 μs and 720 μs respectively.  So I get the wrong answer much much faster ;-)

I think we can make a fast version of matrix operations for uncorrelated errors  (for matrix inverse, see DOI: 10.1016/S0024-3795(01)00273-7), but I haven't had a chance to implement anything yet.  

Uncertainty is also easier to use.  I do need to do a better job of sigfigs.  It should be something like  0.0154(35) for uncertainty 0.0035, but 0.015(4) for uncertainty 0.0036 (see [Benford's Law](https://en.wikipedia.org/wiki/Benford%27s_law)).