reductus/reductus #70: warn user if template modules have changed

pkienzle, 2020-02-14:
It would be nice to notify users on reload which modules have changed since the file was exported.

To do this our stored templates will have to include the datestamp for each module (currently they do not).  When running a template node, all datestamps need to be updated to the versions defined by the instrument (or maybe downgraded if running from an older githash).  Predefined templates should come in without datestamps because that would just be annoying.

This check could be done on the client (saving a round-trip to the server on upload), or the server (making the checking code available to command line applications like regression.py).

bmaranville, 2020-02-14:
In principle this information is all knowable from the git hash, but maybe not terribly easy to recover.

pkienzle, 2020-02-14:
In early versions we were incorporating the python code for the module directly in the template, at least for fingerprinting.  That did have the (dis)advantage of keeping the reduction steps simple.