reductus/reductus #76: minimize metadata sent to the client when loading a directory

pkienzle, 2020-03-10:
The current "metadata" object in reflectometry appears to include the entire dataset.

This makes loading large directories very expensive (e.g., `/candor/202001/nonims2/data`).

bmaranville, 2020-03-10:
Including the data with the metadata was always unnecessary, but sometimes convenient.  If it's no longer convenient we could just strip the data completely though I kind of like the min/max values approach.

pkienzle, 2020-03-11:
Reopening issue.  Set all array values to the min/max when retrieving metadata.  The current "fix" just drops those arrays bigger than 1000 elements.

We may want to go back to delayed loading for the detector and other large block, though for now it seems fast enough even on big directories (ncnrdata/candor/202001/nonims2).

commit e15045ac (2020-03-10): Limit size of returned metadata. Fixes #76.
closed 2020-04-08 by bmaranville