reductus/reductus #78: cannot read property load_file of undefined

pkienzle, 2020-03-10:
﻿
The following error appears when starting reductus in chrome
```
filebrowser.js:344 Uncaught (in promise) TypeError: Cannot read property 'load_file' of undefined
    at Object.handleChecked (filebrowser.js:344)
    at Object.fileinfoUI (fieldUI.js:53)
    at editor.js:295
    at Array.forEach (<anonymous>)
    at editor.js:267
```

... and this one as well:
```
DevTools failed to parse SourceMap: chrome-extension://ekhagklcjbdpajgpjgmbionohlpdbjgc/browser-polyfill.js.map
​```

bmaranville, 2020-03-10:
This suggests that the instrument definition is not being passed to your browser.  I am unable to reproduce this error - do you have instruments defined in config.py?

pkienzle, 2020-03-11:
The instruments are showing up in the instruments menu, so yes.

I've been starting the server using from the repo root using:

    python -m reflweb.run -x

bmaranville, 2020-03-11:
I haven't ever started the server that way, so I'm not sure what's going on.  On all the platforms I've used I just type "reductus" or "reductus --headless" after pip install -e .

bmaranville, 2020-03-11:
Once in a while I do "python3 server_flask.py" from the relfweb directory, which is more or less what the wsgi processes do.

pkienzle, 2020-03-11:
"reductus --headless" should be the same as "python -m reflweb.run -x" given the entry point "reductus=reflweb.run:main" in setup.py.

bmaranville, 2020-03-11:
How can I reproduce this?  What is your python environment and OS?

bmaranville, 2020-03-11:
What are you doing when the error appears - is it on first load of the web app?  What url are you using for the app?

bmaranville, 2020-03-11:
What is in config.py?  Can you post?

pkienzle, 2020-03-12:
In my config.py I have start_path pointing to `nice/reset_server_data`.  Pointing it to an empty directory  instead removes the startup error.

bmaranville, 2020-09-04:
@pkienzle I tried to reproduce this error by setting start_path to the same as you did, but it doesn't seem to happen for me.  Can you reproduce the error still, with the new client?