reductus/reductus #93: NaN in 2d plots

hoogerheide, 2020-08-06:
When mousing over a 2D plot, the third coordinate is always listed as NaN. If it is possible to tie this to the actual value of the pixel, this would be very useful.

bmaranville, 2020-08-31:
This is a good point.  This was set up for the old 2d plotter, but recently everything got shifted to the 2d multi plotter and I suspect that it when it was broken.

bmaranville, 2020-09-01:
Hah - this turns out to have been an issue with the fact that our x-axis goes from high to low.  Other than that i think it's working fine.  (comparisons with xmax and xmin from dims were nonsensical)
Will fix.

bmaranville, 2020-09-01:
fixed by release https://github.com/usnistgov/d3-science/releases/tag/v0.2.7

closed 2020-09-01 by bmaranville