reductus/reductus #95: Template Upload Broken

martintb, 2020-09-03:
The "upload_template" menu item in latest plotters_filebrowsers branch doesn't appear to do anything. The filechooser menu 
appears but after choosing a file to upload,  nothing happens. There don't appear to be any errors thrown in the flask terminal or js console though.

bmaranville, 2020-09-03:
I think there is an issue where it is suppressing errors, so it is silently failing.  Please send me the template so I can check.

martintb, 2020-09-04:
Here's the template: [ncnr.sans.rlj.json.zip](https://github.com/reductus/reductus/files/5174966/ncnr.sans.rlj.json.zip)

The template works fine if it's manually placed in the sansred/templates directory so I believe it's is well-formed.

bmaranville, 2020-09-04:
It is not matching the "signature" of a bare template, which at the moment is defined as "{\n" (see https://github.com/reductus/reductus/blob/plotters_filebrowser/web_gui/webreduce/js/reload.js#L2 )  The reasoning was that all templates downloaded from the client have a newline after the opening bracket, so I thought I would include that in the signature.  Otherwise the file signature of a JSON file is just the single "{" character, which I suppose is fine too.  I will make the change.  I should make the error bubble up too... at the moment it is just showing up in the console as "signatures.find(...) is undefined" which is not that helpful.

closed 2020-09-04 by bmaranville