reductus/reductus #99: parallel processing

bmaranville, 2020-10-12:
If you have a function that processes many datasets (sorting, etc) the result of the calculation is cached as a single bundle with the fingerprint determined by the files chosen.  If functions can be identified as "parallelizable", they could be traced back along the graph to the source (a list of files to be loaded, typically) and the segment of the graph that is made up of parallelizable operations can be calculated for each input separately, where each result then has a unique fingerprint.
Then, when a new file is added to the list, or a file is removed from the list, the recalculation time is small as only the new file needs to be run through the parallel calculations in the first case, or a subset of the already cached calculations needs to be returned in the second case.
This can also mitigate the need for specialized parallel loaders - all loaders could then be marked as parallelized and handled in this way.
If, additionally, a preflight operation is added to the API then the client can get a list of fingerprints that would result from a requested template calculation, so that they could request just the ones that are not already cached on the client side (and the progress meter in the client could reflect the true number of calculations that need to be done, rather than always just being "1")
Additionally, updates to the function signatures on the server side would then automatically cause updates in the client data, as the fingerprints would change, without a client refresh.
It's possible that the client wouldn't have to do any fingerprinting at all anymore, and could rely on the fingerprints returned from the server as keys for the client cache (preflighting every request).  This would simplify the client code.

bmaranville, 2020-10-12:
@pkienzle I made a version of this that operates on the client side only (it creates N copies of the template for N files, along a graph subset of connected parallelizable functions, and then requests them serially) and it seemed to work pretty well, but it still runs into the issue that the results will be cached as a monolithic bundle as soon as the calculation reaches a non-parallelizable operation on the server side.  This will all be very useful when the VSANS people are looking at their hundreds of files per experiment, which need to be sorted and categorized.

pkienzle, 2020-11-21:
A different hash and a different cache for each file in a bundle? Then dispatch each one separately. You don't even need to mark the nodes specially since we already have the info: the input* notation indicates that the particular node requires all the inputs to produce an output.

You would need a synchronization mechanism. For example, when you are dividing all frames by the same detector efficiency, you only want that detector efficiency path to be executed once even though each input depends on it separately. So you need to queue the work that is not already queued, then have some mechanism to wake up the process when all its prerequisites are satisfied.

You also want to pay attention to data locality.  Since operations can occur on large data sets with intermediate values not cached you want the inherently serial steps spooled out to the different workers according to the data they are working with rather than the operations they are performing.

I wonder if we could use dask as the execution back end, and let it sort out these concerns?

bmaranville, 2020-11-23:
I don't think we have to do any particularly complicated orchestration - your calculation routine already stores all the intermediate steps, so if it needs an input value and will use it more than once it will use the stored value the second time.  This is not multi-threaded or multiprocessing-enabled, so everything for a calculation runs in a single thread.  There are multiple processes available for running calculations, for load-balancing, but one calculation is always one thread.