reductus/reductus #101: make reload/upload into menu choices rather than input boxes

pkienzle, 2020-10-17:
At least on chrome, clicking on the upload template and reload exported boxes (paper clip + blank line) immediately pops up a file dialog, which on selection immediately exits the sidebar.  Therefore the text box is unused other than for triggering the dialog, and never shows the name of the file selected.

Furthermore, I find myself clicking the "upload" icon on the right to no avail.

I suggest changing it to the same style as the other sidebar menu options like Edit and Download which highlight on hover and act on click.

bmaranville, 2020-10-20:
Both actions are intentionally triggered by the client (closing the menu, and clearing the input) on selection - the menu is closed because it has finished the action (the file actually is reloading at this point) and the input box is cleared to avoid another, subtler error: when you want to upload the same file with the same name again (if you have modified it) then you have to clear the input before you can select that file again.
I used the file input from the UI library just because it is consistent with the HTML standard inputs - it is in fact a file selection input - but I did hide that input on previous versions of the client and I could do it again.  It's a bit awkward to do but it can be done, I think.

bmaranville, 2020-10-20:
Ok, not that awkward!  Closed by 1eb9e9d5efaee051aa86094d11560e821079d954

closed 2020-10-20 by bmaranville