reductus/reductus #103: candor needs rectilinear grids for data display

pkienzle, 2020-11-20:
Consider [this specular file](https://ncnr.nist.gov/ncnrdata/view/nexus-hdf-viewer.html?pathlist=ncnrdata+candor+202011+27891+data&filename=PtonSi2_ktest_highQ2744.nxs.cdr) which has non-linear spacing in detector angle. When plotted in reductus, it assumes angle is linear from min to max, which is highly misleading.

Short term work-around: display x',y' coordinates of the data point in the hover instead of the x,y of the pixel on the graph coordinates. That way at least the aware user can figure out what is going on.

Medium: use x',y' for the tick mark labels on the x- and y-axes rather than linear spacing between lowest and highest values on that axes. Then the user will  be able to see that the axis is transformed.

Long term: set row/column size for the grid according to Δx', Δy' rather than linear spacing, drawing the data point boxes with the appropriate size.

Alternative: rather than drawing a grid, draw a scatter plot with points colored by intensity. You can get fancy by drawing the point radius according to resolution in x and y.

bmaranville, 2020-11-23:
The long-term goal you mention is similar to the approach we tried first for plotting 2d data with jqplot, many years ago, in dataflow (pre-reductus).
It was very very (very?) slow for datasets larger than 128x128 (too may objects to draw individually) and used a lot of memory.  
This could be improved by drawing directly to canvas instead of SVG, but jqplot used canvas not SVG and was *extremely* slow.  Nothing will be as fast as directly setting pixel colors on the backing image like we are doing now, but maybe it could be close.  Might have to use GL libraries, which are not universally supported and I try to stay away.

If you decide to implement use a backing image with a scale+paint to the user view (makes pan very responsive, because you don't do anything except change the start point for painting!) you will have to make potentially awkward choices for where to make row/column boundaries in pixel coordinates.  

On the other hand if you do a direct paint of pixels in the plot window, at full size, with no backing image, you will have to handle both a potentially large number of pixels to draw, and an algorithm for combining multiple cells in the data into a single pixel of the display when you zoom out past 1:1 pixel:data mapping (the canvas image drawing routines do this for you, when you use the backing image method)

bmaranville, 2020-11-23:
I suppose you could make a hybrid approach where it draws directly to the screen pixels if the data cells projected onto the screen are more than 1 pixel, but draw a backing image with all the cells and scale and display it on the screen if the projected data is smaller than one pixel.  What if it is bigger in one dimension but smaller in the other?

bmaranville, 2020-11-23:
There is an intriguing fiddle here http://jsfiddle.net/alnitak/Dpgts/ where they draw a (circle in this case) for each color in the palette, in its own section of a backing canvas, then place+draw circles from the backing canvas to the main canvas.

pkienzle, 2020-11-24:
The non-linear ticks will be good enough for now. That should be a lot easier, and not too time consuming.

I notice the drawImage method allows arbitrary stretch in each direction. It doesn't change the runtime at all when I do that (still 9 ms on my machine for 3240 = 60x54 dots).

bmaranville, 2020-11-24:
There are a lot of affine transformations available, but none of them directly can create the segmented 2d stretching this feature requires!

pkienzle, 2020-11-24:
Using that fiddle you can draw each rectangle with a stretch. Apparently it is fast enough for a dataset with 60 angles (9 ms), or even 600 angles (75 ms).

Better yet, just draw them directly (24 ms):
```javascript
var colours = ['red', 'green', 'blue', 'yellow', 'magenta'];
var n = colours.length;
var d = 10;

// get handles to the on-screen canvas
var canvas = document.getElementById('canvas');
var ctx2 = canvas.getContext('2d');

// create 3000 random points
var points = [];
for (var i = 0; i < 32400; ++i) {
    points.push({
        c: Math.floor(n * Math.random()),
        x: Math.floor(canvas.width * Math.random()),
        y: Math.floor(canvas.height * Math.random()),
        dx: d * Math.random(),
        dy: d * Math.random()
    });
}

// copy points from the offscreen canvas into the on-page canvas
var t0 = Date.now();
for (var i = 0; i < points.length; ++i) {
    var p = points[i];
    ctx2.fillStyle = colours[p.c];
    ctx2.fillRect(p.x, p.y, p.dx, p.dy);
}
var t1 = Date.now();
alert((t1 - t0) + "ms");
```

Can you embed the drawing commands within the svg that d3 uses?

bmaranville, 2020-11-24:
Yes, you can - I am already doing canvas commands in the backend of the 2d plotter (not drawing rectangles at the moment, but setting pixel values in the context).  I was concerned that directly drawing rectangles would be slow, but the acceleration for this type of operation has apparently gotten better!

bmaranville, 2020-11-24:
It will still be slightly slower to calculate the coordinates of the rectangles, but not too bad - since they are still on a crossed grid (only need to calculate once per row and column)

pkienzle, 2020-11-24:
512 x 640 takes ~0.2 s, so a little bit slow. Of that 0.07 s is for rendering fill style, which seems excessive.

This seems like a job for texture mapping: zap the square pixels into a texture, and use a texture lookup on the coordinates of each pixel based on the nonlinear transformation of the pixel centers. Don't know how to do this in svg or canvas though.

bmaranville, 2020-11-24:
The texture lookup operation you are describing is a GL operation, not in the 2d canvas library.  The transformations available in the canvas library are the affine transformations previously discussed, which do not solve this problem.

pkienzle, 2020-11-24:
Have you played with d3 map projections? These are certainly not affine. The question is whether you can provide a transformation function which returns px,py given qx,qy.

bmaranville, 2020-11-24:
It looks like they have an option for rendering to canvas instead of SVG... this would be necessary for our large number of segments.

pkienzle, 2020-11-24:
A possible canvas-based algorithm:
* render the bitmap to an off-screen canvas
* copy each row to a second off-screen canvas with height stretch
* copy each column to a the visible canvas with width stretch
640 + 512 copies may be faster than rendering 640 x 512 rectangles.

bmaranville, 2020-11-30:
I think it would make sense to create a new type of plot, heatChartRectilinear, to support this.  The simplest thing will be to draw rectangles directly on the visible canvas.  The fastest thing might be to draw all the rectangles to a backing canvas at least as big as the visible canvas, then copy the whole thing after scaling onto the visible canvas.  This will make pan/zoom *much* faster, as fast as it is now roughly.  Changing the colormap will be slower, as it will require redrawing all the rectangles, but maybe not too slow.

pkienzle, 2020-12-01:
For scaling to work the smallest row size will have to be at least one pixel. For pathological cases this could result in very large canvases.

bmaranville, 2020-12-01:
It will be somewhat worse than that... in order to avoid artifacts from data boundaries not landing on pixel boundaries, we would probably require the minimum block be 5 pixels (for 20% distortion) or 10 pixels for 10% distortion.
On the other hand, in my crude testing drawing 1024 * 512 rects takes < 1sec, so it's possible for this special case we should just do the dumb easy thing.

pkienzle, 2020-12-01:
Will zoom and pan be fast? Because 1/2 second on zoom will be painful.

bmaranville, 2020-12-01:
Zoom will only be fast if you commit the draw to a "backing" canvas that can be redrawn on the main canvas with different levels of scaling, so the complications above will come into play (making sure the backing canvas is big enough).  For panning, a backing canvas of the same size as the visible canvas is sufficient.

bmaranville, 2020-12-01:
I think for most of our use cases so far, (54 x 100 or so) the slow/dumb method will suffice, and won't be that slow.  The more complete solution is starting to look complex and not really "complete" anyway (still have to compromise somewhere when converting floating-point boundaries to integer pixels)

bmaranville, 2020-12-07:
The most straightforward way to specify the rectangles is to supply N+1 edges for each axis of length N, similar to how e.g. x_min and x_max are the low edge of the lowest data point and the hight edge of the highest data point already.

pkienzle, 2020-12-07:
You want to go 1/2 bin below xmin and 1/2 bin above xmax, as done in [reflred.candor.edges](https://github.com/reductus/reductus/blob/bba5f27a0929efbcebaec27e2b28a05ad019aff7/reflred/candor.py#L701-L713).