reductus/reductus #106: Feature request: report error in main screen

hoogerheide, 2020-12-03:
When something goes wrong with a reduction path, the plotting panel shows a null symbol. The error message sometimes pops up, but sometimes it has to be accessed through the browser development tools, depending on browser and page settings. Users who are not familiar with the development tools are thus unable to troubleshoot their error. This is particularly important for CANDOR reduction where there hasn't been a consensus on array sizes and how broadcasting and masking works yet; mismatched arrays are fairly common at the moment.

One temporary solution, if possible, would be to dump the traceback into the plotting panel window so at least the user can supply this information when asking for help.

bmaranville, 2020-12-07:
There is a bit of compromise being done... there are a class of errors (like data file not loading, etc.) that make the user experience pretty awful if they block the application and wait for the user to click through and dismiss.  
I've been thinking of redoing the error reporting so that it is part of the page (a little bar that pops up from the bottom) instead of using the browser's notification system.  I think it will be more visible, and it will be possible to improve the interactivity.
That said, I don't expect most users to be troubleshooting errors!

pkienzle, 2020-12-07:
You don't need to popup the error if you dump it as text in the plot window.  It is, after all, the "output" from the operation. 

Make sure the text is selectable text so users can paste it in an email. Even better would be to create the email with the template and the traceback. We could do this automatically on the server when there is an error. The spam would provide motivation to fix things, and we wouldn't have to rely on users reporting errors. Still show the error text on the plot area, but no need for traceback.

Warnings are a different story. I can imagine returning a list of all warnings/errors associated with each node id when requesting output from the server.  Then colour the nodes in the dataflow diagram with the severity of the warning. Maybe have the text of the warning pop up below the short description on hover.

Would we want warnings to be written to the output file?