reductus/reductus #113: select_points is broken

hoogerheide, 2020-12-30:
select_points gives a "float is not subscriptable" error. Might be an error in the input type? Template attached.
[template_select_points_broken.json.txt](https://github.com/reductus/reductus/files/5755525/template_select_points_broken.json.txt)

bmaranville, 2021-01-27:
I think this got fixed with an earlier commit... but your example template has values in the select array that are longer than the data coming in.  If I cut them off arbitrarily at point 28, it now works.  Can you verify that the fix is working with this template, and then I can close this issue?
[select_points_working.json.txt](https://github.com/reductus/reductus/files/5881129/select_points_working.json.txt)

hoogerheide, 2021-01-27:
I don't think the input type of select_point is correct, though; if the input has more than one bank, you can't give it a list of lists because it's expecting a single list of ints.

D