reductus/reductus #114: 2D plots don't update

martintb, 2021-01-01:
The 2D plots don't update if you move from one 2D plotted input/output node to another 2D node. The plotter will continue to show the first 2D plot moving from 2D to 2D until you move to another type of data/plot (e.g., 1D or Table). Only then the next 2D input/output chosen will update correctly.

This is occurring both with the latest pulls from the main branch and with the currently running server on reductus.nist.gov. I'm using the _ncnr.sans_ instrument and it appears to happen with anything that load 2D data (SansData).

bmaranville, 2021-01-27:
I will fix this now... sorry I don't see these issues in my email for some reason.

bmaranville, 2021-01-27:
Fixed by b2147cb

closed 2021-01-27 by bmaranville