reductus/reductus #133: Feature Request: Sort by cross-section type

alexander-grutter, 2021-11-16:
I would love to be able to sort and join my scattering cross sections not just by "polarization" or "instrument" but by "spin-flip" and "non spin-flip"

bmaranville, 2021-11-17:
Apparently this has been requested before, and just never finished... #45

bmaranville, 2021-11-17:
Would you want NSF to include half-polarized, i.e. (+,None), (-, None), (None, +), (None, -), or just (+, +) and (-, -)?

alexander-grutter, 2021-11-17:
Yes, I think that NSF must include half-polarized. If there is spin-flip and we're running a half-polarized state, the entire scattering problem is ambiguous and can't be analyzed anyway. So I think it's best to assume we are reducing useable data.

  *   Alex

bmaranville, 2021-11-17:
It seems like the main place this is desired is in the subtraction of backgrounds.  We can add a flag for "average SF backgrounds before subtract" to the polarized background subtraction, which defaults to False.  It seems like an option for "average NSF backgrounds before subtract" may also be useful (also defaults to False)

bmaranville, 2021-11-17:
Given the complexities of join, it seems like we want to do this mixing before the join.  Implemented a solution in #134

closed 2021-11-17 by bmaranville