reductus/reductus #136: Feature request: subtract fitted background (not from fluid reservoir)

bmaranville, 2022-01-05:
For small (magnetic) samples, it would reduce the propagated noise on R if the background was smoothed before subtraction.  One method to do this is to fit to a functional form, as was done for fluid-reservoir on Si reflectometry in #47 

For the smaller samples, the function seems to be  
```y = p0 + p1*x + p2 * exp(-x * p3)```
for Q > critical edge.  A strategy that seems like it might work is 

 * choose the fitting region
 * fit the background in that region
 * make the output of that module be
   * original bkg up to the cutoff, then
   * new functional-form bkg above the cutoff (on a tight grid, maybe with spacing set in the module itself)

pkienzle, 2022-02-09:
How about a smoothing filter?

For some width k fit all points in ± k⋅ΔQ to a polynomial of degree d=3. Replace the point with the fitted value. It's like a variable width Savitsky-Golay filter. This way you still get the benefit of improved statistics from neighbouring points without imposing a model on the background.

bmaranville, 2022-02-10:
True, it is more model-independent to apply a polynomial local filter - but the benefit is correspondingly lower as well.  Looking at the background measurements from PBR and CANDOR, they are very *boring*, yet most of the uncertainty on individual points at high Q is coming from the background subtraction.  We have a priori knowledge that the background does not vary rapidly there, but we are not using this prior knowledge.  Combining multiple points with S-G lowers the uncertainty, combining all the points with a reasonable model lowers it more.

On the other hand, the S-G option is simpler, more explainable, and therefore easier to justify in a paper.  That is a strong argument for using that instead.

pkienzle, 2022-02-10:
To see what could be gained from a correct background model I set up Poisson sampling from B and S = R + B with B ≈ R. I then compared ΔR/R for R = S - B with B perfect (i.e., fitted background ΔΒ ≪ ΔS) and with Β measured. Using a simulated time of 1.0 hours per point I calculated ΔR/R ≈ 7.5 % for B perfect, which is the best that fitted B could achieve. The get the same ΔR/R with measured B requred a simulated tiime of 1.4 hours.
```
B = 1e-8; I = 1e7; t = 3600; n = 1_000_000
R = B; h=1; Io = I*t*h; S=np.random.poisson((R+B)*Io, n)/Io; print(f"fitted B: {std(S-B)/R*100:.2f}% ΔR/R, measured B: {std(S - np.random.poisson(B*Io, n)/Io)/R*100:.2f}% ΔR/R")
```

The 40% time reduction is substantial. I suspect you will get that with a smoothing filter, at least if you measure the background at each point. The question is whether you can undersample at high angle and use your proposed model to reduce uncertainty.

The model: _y_ = _p₀_ + _p₁x_ + _p₂_ exp(–_p₃x_)

The constant _p₀_ can come from any number of sources such as dark counts on the detector or air scatter for fixed slits. I can imagine a linear term _p₁_ due to distance from beam stop. I can imagine a quadratic term (not included) following beam intensity as the slits open. Any idea where the exponential term might come from?

If the radiation field has bumps (e.g. because of details of the shielding or unexpected scatter from beamline components) then you will have small differences from the background model as the detector sweeps through space. Obviously these effects will be subtle or we would have fixed them, but I can imagine a few percent change in background going unnoticed. This is should be okay so long as it is much smaller than typical ΔR.

I also played with R ≪ B, which you can measure if you count long enough. For R = B/10 the saving was 2x (100 hours per point unfitted, 50 hours fitted).  Here ΔR is much smaller, so you will need a much better model of the background. I wouldn't trust the model for tiny R without some very long calibration measurements. 

Background is a non-issue for R ≫ B. Adding in quadrature means that ΔS dominates. In current practice we save time by undersampling the background at low angle. Reductus uses a linear spline (i.e., interpolation) in this case, but it could be replaced with a smoothing filter with no practical effect on uncertainty.