reductus/reductus #140: Add support for other versions of Bruker X-ray file

bmaranville, 2022-02-25:
Some hints for how to interpret the binary fields can be found in https://gsas-ii.readthedocs.io/en/latest/_modules/G2pwd_BrukerRAW.html

bmaranville, 2022-03-03:
In the near term, suggesting that people use PowDLL to convert to version 1.01 of the RAW file is probably a reasonable thing to do (it is likely to preserve fields that aren't defined in the library above), although even powDLL is not able to convert all RAW4 files.

pkienzle, 2022-04-14:
GSAS-II has the BSD attribution clause, which makes using it wholesale kind of inconvenient.

I have a bruker raw reader in [scattio/bruker.py](https://github.com/scattering/scattio/blob/master/python/scattio/bruker.py).

I'm checking for RAW1.01, which according to Brian's code is Bruker RAW ver. 3.

pkienzle, 2022-05-05:
Do we have users that need other versions? Can they provide information about the file format?

bmaranville, 2022-05-07:
Users have asked about other versions but they do not have access to the file format specification.  As far as I know it is not published online.