reductus/reductus #147: negative values not appearing on log scale plots

pkienzle, 2022-06-15:
A user reported that negative values did not appear on the log plot. This makes it difficult to mask off high q values which are sometimes negative due to background subtraction.

bmaranville, 2022-06-15:
via email:
>  min (R + ΔR)/2 as the floor of the log region works pretty well. It protects against something like -1e-40 as the largest magnitude negative value, which would otherwise give an unusable range on the log portion of the plot. Or 1e-40 as the smallest value with no negative values.

This would work for values with error bars, but otherwise it's a little hard to guess where to put the linear part.

pkienzle, 2022-06-15:
All values have uncertainty, we just need to specify it. 8-)

For those cases where uncertainty is unknown we could use a 95% interval. Something like (untested):
```javascript
if (values.length > 3) { // don't die on small datasets
   // From eranws.github.com
   // https://stackoverflow.com/questions/46622486/what-is-the-javascript-equivalent-of-numpy-argsort/65410414#65410414
   // with mods to find the 95% range of absolute values.
   let decor = (v, i) => [Math.abs(v), i];          // set index to value
   let undecor = a => a[1];               // leave only index
   let argsort = arr => arr.map(decor).sort().map(undecor);
   let index = argsort(values);
   // trim the 95% interval by one on each end to guarantee some robustness on small datasets
   let vmax95 = values[index[Math.floor(values.length*0.975)-1]];
   let vmin95 = values[index[Math.floor(values.length*0.025)+1]];
   let range = vmax95-vmin95;
   let logfloor = vmin95 - 0.05*range;
}
```