reductus/reductus #149: Use external nxsunit package

pkienzle, 2022-07-13:
I created a new repo on github ([scattering/nxsunit](https://github.com/scattering/nxsunit)).

Test using: `pip install git+https://github.com/scattering/nxsunit.git`

I'll put it on pypi after it is [reviewed](https://github.com/scattering/nxsunit/commit/ce71e114bfce1c1e6e7411b62b0670e630be32d3).

bmaranville, 2022-07-13:
Where will this be used in reductus?

pkienzle, 2022-07-13:
I found that I needed the nexus read/convert in yet another project so I created a package for it. 

This replaces `dataflow.lib.unit` used by `reflred.nexusref`, `sansred.loader` and `vsansred.loader`, and `reflred.nexusref.data_as`. It is probably a drop-in replacement, but it'll require testing. We can be conservative and leave the existing `unit.py` in place, pulling in a newer version from upstream if we need it.

I started from the version on scattio which handles temperature conversion properly, and added some updates from sasview (ignore units if they are not given in the file rather than throwing an error). I added simplified interfaces for `convert(value, from, to)` and `u=conversion(from, to);u(value)`. I pulled `reflred.nexusref.data_as` into nxsunit since I use both of them to get data out of nexus files on a couple of projects.