reductus/reductus #165: SANS: `Abs Scale` module input parameters need to be rearranged

krzywon, 2025-07-16:
The `empty trans` data is used in multiple places, but is called first in the `Abs scale` module. From a flow-chart standpoint, to ensure minimal crossing data paths, this should be the last input for the module.

To prevent potential backwards-compatility issues, a new module should be created, with rearranged inputs, that calls the existing module.

bmaranville, 2025-07-16:
I think the execution framework passes all arguments to reduction steps as a `kwargs` dict (all inputs are named), rather than as positional arguments, so you may not  have to create a new module (reordering the inputs should be ok).

krzywon, 2026-10-27:
This was fixed in #210

closed 2026-10-27 by krzywon